openenergymonitor/EmonESP #22: Prefer radio button than check button when selecting network

minida28, 2017-05-20:
I setup a wifi repeater at home and the broadcasted SSID from both routers (same SSSID) are showed up on emonESP network selection screen.

As shown in the below screenshot, I can select both SSIDs. Not sure why, maybe it is because both SSID are identical.

Not a big deal, but it would be better if the UI implements radio button rather than check box, so only one network that can be selected at a time.

[image]

glynhudson, 2017-05-20:
Good idea!

glynhudson, 2017-05-20:
done 👍

closed 2017-05-20 by glynhudson

minida28, 2017-05-20:
Hi @glynhudson 

I've re-tested it, the button type has changed to radio.
But the strange thing is I can still select multiple networks!

[image]

minida28, 2017-05-20:
I think I have found the issue.
Chrome developer tool throws this error:
```
Uncaught ReferenceError: networkcheckboxes is not defined
    at HTMLInputElement.networkSelect (config.js:383)
```
Adding this line just above line 383 on config.js file seems solve the problem:

```
var networkcheckboxes = document.getElementsByClassName("networkcheckbox");
```

glynhudson, 2017-05-20:
Thanks, done.

commit bdeb5f3c (2017-05-20): change checkbox to radio box fixes #22
commit 67628429 (2017-05-20): fix #22