openenergymonitor/EmonESP #30: Includes missing

emtemt, 2017-09-22:
Hi
Just downloaded and will not compile
webserver.h has 2 includes not in git files
<ESPAsyncTCP.h>  <ESPAsyncWebServer.h>

pieman64, 2017-09-22:
@emtemt they are included in the releases, just not in the master.

brucemiranda, 2017-10-30:
I have this same issue. And sorry I don't understand what included in the releases means? 

What do I need to add to get the project to compile. I tried added ESPAsyncTCP and ESPAsyncWebServer as libraries and then I get whole host of other errors.

glynhudson, 2017-10-31:
You can download the required librarys from here:

https://github.com/me-no-dev/ESPAsyncTCP
https://github.com/me-no-dev/ESPAsyncWebServer

I highly recomend using [PlatformIO](https://platformio.org) to compile. Libraries will be automatically installed at the correct version. See instructions in readme and our platformio guide:

https://guide.openenergymonitor.org/technical/compiling

closed 2017-10-31 by glynhudson

brucemiranda, 2017-10-31:
I added those two libraries to the Arduino IDE and then I get a bunch of new errors. 

`Arduino: 1.8.5 (Windows 10), Board: "Adafruit HUZZAH ESP8266, 80 MHz, 115200, 4M (3M SPIFFS)"

sketch\config.cpp: In function 'void EEPROM_write_string(int, int, String)':

sketch\config.cpp:117:22: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

     if (i<val.length()) {

                      ^

sketch\emoncms.cpp: In function 'void emoncms_publish(String)':

sketch\emoncms.cpp:47:35: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

   for (int i = 0; i < data.length(); ++i){

                                   ^

In file included from <home dir>\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:444:0,

                 from sketch\web_server.h:31,

                 from sketch\ota.cpp:28:

<home dir>\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:67:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator ++(int i) { _count++; }

          ^

<home dir>\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:68:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator --(int i) {  if (_count > 0) { _count--; } ;  }

          ^

In file included from <home dir>\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:444:0,

                 from sketch\web_server.h:31,

                 from <home dir>\Arduino\EmonESP-master\src\src.ino:29:

<home dir>\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:67:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator ++(int i) { _count++; }

          ^

<home dir>\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:68:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator --(int i) {  if (_count > 0) { _count--; } ;  }

          ^

In file included from <home dir>\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:444:0,

                 from sketch\web_server.h:31,

                 from sketch\web_server.cpp:31:

<home dir>\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:67:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator ++(int i) { _count++; }

          ^

<home dir>\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:68:10: warning: unused parameter 'i' [-Wunused-parameter]

[... 282 lines cut ...]
                              ^

In file included from <home dir>\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:27:0:

C:\Users\The Mirandas\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:317:18: error:   initializing argument 2 of 'void tcp_accept(tcp_pcb*, tcp_accept_fn)' [-fpermissive]

 void             tcp_accept  (struct tcp_pcb *pcb, tcp_accept_fn accept) ICACHE_FLASH_ATTR;

                  ^

<home dir>\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp: At global scope:

<home dir>\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:899:6: warning: unused parameter 'err' [-Wunused-parameter]

 long AsyncServer::_accept(tcp_pcb* pcb, long err){

      ^

exit status 1
Error compiling for board Adafruit HUZZAH ESP8266.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

brucemiranda, 2017-10-31:
When I add those two libraries I get a lot of other errors. 

`Arduino: 1.8.5 (Windows 10), Board: "Adafruit HUZZAH ESP8266, 80 MHz, 115200, 4M (3M SPIFFS)"

sketch\config.cpp: In function 'void EEPROM_write_string(int, int, String)':

sketch\config.cpp:117:22: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

     if (i<val.length()) {

                      ^

sketch\emoncms.cpp: In function 'void emoncms_publish(String)':

sketch\emoncms.cpp:47:35: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

   for (int i = 0; i < data.length(); ++i){

                                   ^

In file included from C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:444:0,

                 from sketch\web_server.h:31,

                 from sketch\ota.cpp:28:

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:67:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator ++(int i) { _count++; }

          ^

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:68:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator --(int i) {  if (_count > 0) { _count--; } ;  }

          ^

In file included from C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:444:0,

                 from sketch\web_server.h:31,

                 from C:\Users\ThePC\Documents\Arduino\EmonESP-master\src\src.ino:29:

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:67:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator ++(int i) { _count++; }

          ^

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:68:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator --(int i) {  if (_count > 0) { _count--; } ;  }

          ^

In file included from C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:444:0,

                 from sketch\web_server.h:31,

                 from sketch\web_server.cpp:31:

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:67:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator ++(int i) { _count++; }

          ^

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:68:10: warning: unused parameter 'i' [-Wunused-parameter]

[... 282 lines cut ...]
                              ^

In file included from C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:27:0:

C:\Users\ThePC\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:317:18: error:   initializing argument 2 of 'void tcp_accept(tcp_pcb*, tcp_accept_fn)' [-fpermissive]

 void             tcp_accept  (struct tcp_pcb *pcb, tcp_accept_fn accept) ICACHE_FLASH_ATTR;

                  ^

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp: At global scope:

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncTCP\src\ESPAsyncTCP.cpp:899:6: warning: unused parameter 'err' [-Wunused-parameter]

 long AsyncServer::_accept(tcp_pcb* pcb, long err){

      ^

exit status 1
Error compiling for board Adafruit HUZZAH ESP8266.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

jeremypoulter, 2017-10-31:
You need to be building with the Git version of the ESP8266 core, see https://github.com/esp8266/Arduino#using-git-version

brucemiranda, 2017-10-31:
Ok that seemed to at least compiled. But still quite a lot of warnings. 

`Build options changed, rebuilding all
sketch\config.cpp: In function 'void EEPROM_write_string(int, int, String)':

sketch\config.cpp:117:22: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

     if (i<val.length()) {

                      ^

sketch\emoncms.cpp: In function 'void emoncms_publish(String)':

sketch\emoncms.cpp:47:35: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]

   for (int i = 0; i < data.length(); ++i){

                                   ^

In file included from C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:444:0,

                 from sketch\web_server.h:31,

                 from sketch\ota.cpp:28:

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:67:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator ++(int i) { _count++; }

          ^

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:68:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator --(int i) {  if (_count > 0) { _count--; } ;  }

          ^

In file included from C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:444:0,

                 from sketch\web_server.h:31,

                 from C:\Users\ThePC\Documents\Arduino\EmonESP-master\src\src.ino:29:

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:67:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator ++(int i) { _count++; }

          ^

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:68:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator --(int i) {  if (_count > 0) { _count--; } ;  }

          ^

In file included from C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:444:0,

                 from sketch\web_server.h:31,

                 from sketch\web_server.cpp:31:

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:67:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator ++(int i) { _count++; }

          ^

C:\Users\ThePC\Documents\Arduino\libraries\ESPAsyncWebServer\src/AsyncWebSocket.h:68:10: warning: unused parameter 'i' [-Wunused-parameter]

     void operator --(int i) {  if (_count > 0) { _count--; } ;  }
[... 464 lines cut ...]

     if (qos < 0 || qos > 1) {

               ^

D:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\spiffs\spiffs_hydrogen.c: In function 'SPIFFS_buffer_bytes_for_filedescs':

D:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\spiffs\spiffs_hydrogen.c:16:49: warning: unused parameter 'fs' [-Wunused-parameter]

 u32_t SPIFFS_buffer_bytes_for_filedescs(spiffs *fs, u32_t num_descs) {

                                                 ^

D:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\Esp.cpp: In member function 'bool EspClass::eraseConfig()':

D:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\Esp.cpp:399:10: warning: unused variable 'ret' [-Wunused-variable]

     bool ret = true;

          ^

Archiving built core (caching) in: C:\Users\THEMIR~1\AppData\Local\Temp\arduino_cache_551026\core\core_esp8266com_esp8266_generic_CpuFrequency_80,CrystalFreq_26,FlashFreq_40,FlashMode_dio,UploadSpeed_115200,FlashSize_512K64,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____d8fd2f18ef2457db5d6b1c756af3e43c.a
Sketch uses 428107 bytes (98%) of program storage space. Maximum is 434160 bytes.
Global variables use 43164 bytes (52%) of dynamic memory, leaving 38756 bytes for local variables. Maximum is 81920 bytes.
`

1danielsan, 2017-11-17:
Hi All, it's been 4 month i try to compile this project without success.
I tried with arduino, adding all the library but nothing (at this moment the error is about eeprom.h)

I finally tried with platformio but thats the output!:

#
# Project Configuration File
#
# A detailed documentation with the EXAMPLES is located here:
# http://docs.platformio.org/en/latest/projectconf.html
#

# A sign `#` at the beginning of the line indicates a comment
# Comment lines are ignored.

# Simple and base environment
# [env:mybaseenv]
# platform = %INSTALLED_PLATFORM_NAME_HERE%
# framework =
# board =
#
# Automatic targets - enable auto-uploading
# targets = upload

[platformio]
env_default = emonesp
data_dir = src/data

[common]
version = -DBUILD_TAG=2.2.3
lib_deps = PubSubClient@2.6, ESPAsyncWebServer@1.1.0

[env:emonesp]
platform = espressif8266
framework = arduino
board = esp12e
lib_deps = ${common.lib_deps}
src_build_flags = ${common.version} -DWIFI_LED=0 -DENABLE_DEBUG
monitor_baud = 115200

[env:emonesp_fast]
# Upload at faster baud reate 50s upload reduced to  20s :-)
platform = espressif8266
framework = arduino
board = esp12e
lib_deps = ${common.lib_deps}
src_build_flags = ${common.version} -DWIFI_LED=0 -DENABLE_DEBUG
upload_speed=921600
monitor_baud = 115200

[env:emonesp_ota]
platform = espressif8266
framework = arduino
board = esp12e
upload_port = emonesp.local
lib_deps = ${common.lib_deps}
src_build_flags = ${common.version} -DWIFI_LED=0 -DENABLE_DEBUG
monitor_baud = 115200

[env:emonesp_spiffs]
platform = espressif8266
framework = arduino
board = esp12e
lib_deps = ${common.lib_deps}
src_build_flags = ${common.version}
upload_flags = --spiffs
monitor_baud = 115200

[env:emonesp_deploy]
platform = espressif8266
framework = arduino
board = esp12e
lib_deps = ${common.lib_deps}
src_build_flags = !(test -z $TRAVIS_TAG && echo '${common.version}') || echo '-DBUILD_TAG='$TRAVIS_TAG

[env:emonesp01]
platform = espressif8266
framework = arduino
board = esp01
lib_deps = ${common.lib_deps}
src_build_flags = ${common.version} -DESP01 -DENABLE_DEBUG
monitor_baud = 115200

# http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version
# Staging Arduino framework and common lib version
[env:emonesp_staging]
platform = espressif8266_stage
board = esp12e
framework = arduino
lib_deps = ${common.lib_deps}
src_build_flags = ${common.version} -DWIFI_LED=0 -DENABLE_DEBUG
upload_port = emonesp.local
monitor_baud = 115200

# staging Arduino framwork and latest libs
[env:emonesp_staging_libs]
platform = espressif8266_stage
board = esp12e
framework = arduino
lib_deps = https://github.com/knolleary/pubsubclient, https://github.com/me-no-dev/ESPAsyncWebServer.git, https://github.com/me-no-dev/ESPAsyncTCP.git
src_build_flags = ${common.version} -DWIFI_LED=0 -DENABLE_DEBUG
upload_port = emonesp.local
monitor_baud = 115200

I don't know how to copy the compile log file! copy/paste doesnt work.

Where is the problem?!

glynhudson, 2017-11-17:
Compiling with platformIO is very straightforward all the libraries are automatically installed. See our PlatformIO user guide:

https://guide.openenergymonitor.org/technical/compiling/

If you just want to use EmonESP you could upload a pre-compiled .bin from the releases page: https://github.com/openenergymonitor/EmonESP/releases

You can upload using esptool:

`esptool.py write_flash 0x000000 firmware.bin 0x300000 spiffs.bin`