openenergymonitor/EmonESP #33: First Boot : authentication required

Tokn59, 2018-01-12:
Hello, I flashed the ESP8266 according to instructions and also uploaded the files in "data". ESP starts and displays the AP emonESPxxxxxxx. I connect and go to 192.168.4.1. It then asks me user id and password.

What should I enter here ? I was expecting a first page where I need to sleect a WIFI access point and password.

Thanks for your support !

Tony

pieman64, 2018-01-14:
@Tokn59 make sure Serial Monitor is closed when you try to access the AP and then it will not ask for a username and password.

Tokn59, 2018-01-15:
Hello, thank you for the suggestion ! I have reflashed the ESP and uploaded the files. Then I connected the ESP to power (not Pc, no Serial Port, just a 5V DC power adapter) After a while the WIFI AP starts up as before emonESP_2063066. Then indeed, no user ID or password are requested but when browsing to http://192.168.4.1 I get the screen (see screenshot) but it keeps saying "Loading, please wait .." forever ... Any more thoughts ?
[image]

pieman64, 2018-01-15:
@Tokn59 think you might have closed the issue by mistake.
Which version of the ESP8266 Arduino core are you using as there was a hastily fixed bug in 2.4.0 but the fix is only in the master branch. You might want to drop back down to 2.4.0-rc2 to see if that helps and if it does then try the master.

Tokn59, 2018-01-15:
I have no idea .. How can I see the version of the ESP8266 Arduino CORE in the IDE ?

Tokn59, 2018-01-15:
I did notice one difference in the instructions, I flashed for ESP8266-12E (which I am using) and the doc said to use Generic ESP8266 board. I rcomplied with that setting and uploaded the files from data but now the emonESP-XXXX doesn't even come up ...

pieman64, 2018-01-15:
Are you flashing the bin files or the flashing from the ino sketch.

Paste the contents of File, Preferences, additional boards url and I should be able to tell you which core you are using.
There is actually a note that states the bin file was compiled with 2.4.0-rc2.

Tokn59, 2018-01-15:
THANK YOU SO MUCH  for trying to help me !

I am trying to flash tyhe src.ino ...

Contents of Preferences : http://arduino.esp8266.com/stable/package_esp8266com_index.json

pieman64, 2018-01-15:
I have just been through the full process again and it brought up the web page as expected.
From the json file you have the core will almost certainly be 2.4.0 as that is the latest official release. If for some reason your IDE didn't automatically update then your core will be 2.3.0.
Change the url to https://github.com/esp8266/Arduino/releases/download/2.4.0-rc2/package_esp8266com_index.json to see if this helps.

Tokn59, 2018-01-15:
Unfortunately exact same result ... Any specific settings in the IDE needed ?

Mine are : CPU freq : 80 Mhz, Flash Size 4M ...

pieman64, 2018-01-15:
Nothing specific in the IDE. You could try uploading the "data" to the ESP again and watch to see that it is shown as completed in the IDE.

pieman64, 2018-01-15:
Did you restart the IDE after changing the boards url?

Tokn59, 2018-01-15:
Yes I did indeed .... Could you plesase share the exact URL where you downloaded src.ino and the other file ... I already saw 2 different versiosn and want to make sure I use the same source and files as you do ...

pieman64, 2018-01-15:
I do a fairly manual process rather than git clone etc. I use the green "Clone or Download" button at https://github.com/openenergymonitor/EmonESP to download the zip file. Put the required files in sketchbook, upload the data and flash the ino etc.

Tokn59, 2018-01-15:
Ok, thank You !!! I will restart from scratch with this one tommorrow evening. Thanks a lot for your help !!!

Kind Regards, Tony

pieman64, 2018-01-16:
Make sure you also do a boards manager revision to roll back to 2.4.0-rc2 as some systems don't update automatically.

Tokn59, 2018-01-19:
Sorry for the delay in resopnding but been away for a few evenings. I took your advise and restarted from scratch. Downloaded from the github link you sent me and opened src.ino. Flashed the ESP8266 and data.

One thing I noticed when uploading data .. should my setting be "ck" or "nodeMCU" in the Arduino IDE ? This is the upload info I get : 

Build options changed, rebuilding all
Archiving built core (caching) in: C:\Users\tonyk\AppData\Local\Temp\arduino_cache_948593\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadSpeed_115200,FlashSize_4M3M_5a391dd20adb797b839917aefa7beb31.a
Sketch uses 381159 bytes (36%) of program storage space. Maximum is 1044464 bytes.
Global variables use 45852 bytes (55%) of dynamic memory, leaving 36068 bytes for local variables. Maximum is 81920 bytes.
Uploading 385312 bytes from C:\Users\tonyk\AppData\Local\Temp\arduino_build_56802/src.ino.bin to flash at 0x00000000
................................................................................ [ 21% ]
................................................................................ [ 42% ]
................................................................................ [ 63% ]
................................................................................ [ 84% ]
.........................................................                        [ 100% ]
[SPIFFS] data   : C:\Users\tonyk\Dropbox\ESP8266 Projects\EmonESP-master\EmonESP-master\src\data
[SPIFFS] size   : 3052
[SPIFFS] page   : 256
[SPIFFS] block  : 8192
/config.js

/home.html

/lib.js

/style.css

[SPIFFS] upload : C:\Users\tonyk\AppData\Local\Temp\buildbb8c4a58b1a05288db14d90d185da73d.spiffs/src.spiffs.bin
[SPIFFS] address: 0x100000
[SPIFFS] reset  : nodemcu
[SPIFFS] port   : COM7
[SPIFFS] speed  : 115200

Uploading 3125248 bytes from C:\Users\tonyk\AppData\Local\Temp\buildbb8c4a58b1a05288db14d90d185da73d.spiffs/src.spiffs.bin to flash at 0x00100000

................................................................................ [  2% ]

................................................................................ [  5% ]

................................................................................ [  7% ]

................................................................................ [ 10% ]

................................................................................ [ 13% ]

................................................................................ [ 15% ]

................................................................................ [ 18% ]

................................................................................ [ 20% ]

................................................................................ [ 23% ]

................................................................................ [ 26% ]

................................................................................ [ 28% ]

................................................................................ [ 31% ]

................................................................................ [ 34% ]

................................................................................ [ 36% ]

................................................................................ [ 39% ]

................................................................................ [ 41% ]

................................................................................ [ 44% ]

................................................................................ [ 47% ]

................................................................................ [ 49% ]

................................................................................ [ 52% ]

................................................................................ [ 55% ]

................................................................................ [ 57% ]

................................................................................ [ 60% ]

................................................................................ [ 62% ]

................................................................................ [ 65% ]

................................................................................ [ 68% ]

................................................................................ [ 70% ]

................................................................................ [ 73% ]

................................................................................ [ 76% ]

................................................................................ [ 78% ]

................................................................................ [ 81% ]

................................................................................ [ 83% ]

................................................................................ [ 86% ]

................................................................................ [ 89% ]

................................................................................ [ 91% ]

................................................................................ [ 94% ]

................................................................................ [ 96% ]

................................................................................ [ 99% ]

............                                                                     [ 100% ]

Once done I unplug from the Pc and connet to a 5V power source. On my iPhone I get : (see screenshots attached) ... No clue what is going wrong !

When I leave it connected to the Pc and look at the debug monitor it does seem to work as it scans for networks : 

EmonESP 751935
Firmware: BUILD_TAG
Connecting to SSID: ; =!=#>A}
 
Try Again...
 
Try Again...
 
Try Again...
 
Try Again...
 
Try Again...
Starting APScan: 10 networks found
AP IP Address: 192.168.4.1
Server started
[image]
[image]
[image]

Tokn59, 2018-01-19:
Looking a bit further I see that I had to manually download some additional files not in the github package. One of them is : ESPAsyncTCP.h .. Maybe I am using a different one than you ?

pieman64, 2018-01-19:
Yes I have a note in my sketch that there are 3 additional libraries required i.e. PubSubClient, ESPAsyncTCP and ESPAsyncWebServer. Presumably you installed the full libraries.
The cku /nodemcu is simply the method the ESP uses for reset, bare ESP's generally cku and NodeMCU / WeMos use nodemcu.
Let me plug mine in again and compare what I see to your screenshots.

pieman64, 2018-01-19:
ESP boots the same as yours:
```
EmonESP 13580552
Firmware: BUILD_TAG
Starting APScan: 6 networks found
AP IP Address: 192.168.4.1
Server started
```

Tokn59, 2018-01-19:
Could you possibly mail me those file so I am absolutely sure I have the exact same conditions as you ?

pieman64, 2018-01-19:
I have created 2 zip files for you. One is for the 3 extra libraries and you need to ensure you IDE is closed before you do any of this. Cut and paste the 3 libraries you currently have for safekeeping. Don't use copy, must be cut. Then unzip my 3 into your libraries directory.

The other zip is the src directory with src.ino, about a dozen .h and .cpp files plus the data directory with webserver files. Put this in sketchbook. Use file upload for the data and normal flash for src.

Where do you want them sending to?

Is there any reason you are using a phone to connect to the AP etc rather than a PC.

I am using a PC and I don't see the asyncesp screen. I think it's probably the steps you are doing rather than bad code.

Tokn59, 2018-01-19:
THANK YOU !!!!! You can send them to my email address : tony.knors@live.com

pieman64, 2018-01-21:
@glynhudson do you have any idea why @Tokn59 is seeing the authentication request shown below when trying to set up EmonESP? I have sent him copies of all my files and he has done a fresh install on a clean PC but still gets the authentication issue.

[image]

sindudas, 2018-01-24:
Maybe there is some data on EEPROM, that EmonESP uses as User/Pass for Web credentials?

It should verifiy if data is Ok or corrupted/not valid. And if not, reset all EEPROM data.

Try to force ResetEEPROM() on setup(), and then reflash without it.

emtemt, 2018-01-29:
Hi

I have seen this authentication screen. Solved by erasing ESP as per the instructions in the platform io section.  I also had problems using the Arduino IDE. I switched to using platform which worked first time.

Ian

pieman64, 2018-01-29:
Thanks @sindudas and @emtemt 
Clearing the ESP of everything finally sorted it for @Tokn59

Tokn59, 2018-01-29:
Indeed .. Full ERASE of the FULL ESP !

pieman64, 2018-01-29:
@glynhudson I think the full ESP erase fix should be in the general section of the docs because users like me that don't like Platform IO will not read that section.

glynhudson, 2018-01-29:
Good to hear you have managed to solve the issue. Thanks a lot to everyone who helped troubleshoot. We highly recomend using platformio to compile and upload emonesp. The erase is documented in the troubleshooting section: https://github.com/openenergymonitor/EmonESP#erase-flash

closed 2018-01-29 by glynhudson

pieman64, 2018-01-29:
@glynhudson but that section comes under **Option 1: Using PlatformIO** so Arduino IDE users will not read it and it's not specific to platformio.

glynhudson, 2018-01-29:
Ah I see, I've moved the troubleshooting section to the general section of the readme at the end. Since it applies to both platformio and Arduino IDE

https://github.com/openenergymonitor/EmonESP/commit/f1db157459ed9b989f3d9851c4101f845fa2a59e