openenergymonitor/EmonESP #34: HTTPS + Not using emoncms.org + /emoncms = Doesn't work

sindudas, 2018-01-24:
If I use https://example.com/emoncms as my EmonCMS server, I can't use HTTPS.

It's because when trying to connect to Server, uses "example.com/emomcms" has "host", and 
`client.connect(host, httpsPort)` fails.
If I put "/emoncms" at the beginning of "e_url", and leave "host" as "example.com" it works!

Maybe an input box to put the server url, will help.

glynhudson, 2018-01-24:
I can confirm we have been able to replicate the issue. Thanks a lot for flagging this up. I'll take a look

mabi, 2018-03-15:
I am addressing this issue in my branch [dev-issue34](/mabi/EmonESP/tree/dev-issue34) still work in progress.

mabi, 2018-03-15:
@sindudas @glynhudson can you please build and test my branch? I don't have the hardware at hand.

sindudas, 2018-03-16:
Flash erase , and then upload firmware and spiffs. Configured and published data through HTTPS

Good job! Thanks

mabi, 2018-03-16:
@sindudas thank you for the test! I will update the docs with new screenshot and create a pull request, then we can close the issue

glynhudson, 2018-03-28:
Nice work, please could you submit a PR for this. Happy to merge. Thanks a lot

closed 2018-03-28 by glynhudson