openenergymonitor/EmonESP #40: MQTT no data

ArashiCZ, 2018-04-05:
Hello,

I just installed brand new EmonTX v3 with EmonESP Huzzah.
I'm having issue with recieving data through MQTT. 
According to MQTTFX, the emonesp is connected to the server, but there is no log recieving any data at all. The base-topic is set to: emon/emonesp
MQTTFX is subscribed to: emon/emonesp/CT1
[image]

I'm getting values on the emonesp webserver, there are no units nex to values by the way.
[image]

Can someone help me with this?
Thanks

glynhudson, 2018-04-05:
All setting looks correct. Are you using an emonPi?

ArashiCZ, 2018-04-05:
Hello,

Well, I'm stupid.
I just tried lower case ct1 - ct3 and it works.
I'm using RPi3 running Node Red as Home Automation for my house.
By the way, I'm one of the guys working on the Revolution pi.

Can you look on the units next to the values?

Thanks a lot

glynhudson, 2018-04-05:
Great!
Wow, fantastic. Revolution Pi looks like a really tidy product, nice work. 
Units can be added in Emoncms graphs.

closed 2018-04-05 by glynhudson