openenergymonitor/EmonESP #42: Have the possibility to define on wich port is running emoncms server

E-NINA, 2018-09-11:
Hello,

I'm running emoncms not on the defult 80 port, i'm using 8080, It will be usefull have the possibility to define on wich port is running emoncms server.

I'm just setting ip adress plus door and can't connect

jeremypoulter, 2018-09-12:
If you are using plain HTTP you can just append :8080 to the end of the server name.

E-NINA, 2018-09-12:
Hello,

I've already try that but no connection maybe is something wrong with my built.
I've built with arduino IDE to a D1 mini board.
I get information from serial port but seems http url isn't passing correctly.
below screen shot of esp output:
Is supossly correct display only "/input/post.json?json=......"?

[image]

sindudas, 2018-09-13:
Hello,

I flashed a new Wemos D1 Mini to test connection to server on port 8080.

> Got 'CT1:3935,CT2:325,T0:20.5'
> /emoncms/input/post.json?json={CT1:3935,CT2:325,T0:20.5,psent:1,psuccess:0,freeram:22952}&node=EmonOUT&apikey=43801fb554f8331431bb913bb92b98a9
> Plain old HTTP
> ok
> Got 'CT1:3935,CT2:325,T0:20.5'
> /emoncms/input/post.json?json={CT1:3935,CT2:325,T0:20.5,psent:2,psuccess:1,freeram:23056}&node=EmonOUT&apikey=43801fb554f8331431bb913bb92b98a9
> Plain old HTTP
> ok

It worked for me adding ":8080" to "Emoncms Server", and without HTTPS.

Maybe there is connection issues from EmonESP to EmonCMS server.

E-NINA, 2018-09-13:
Hello,
My emoncms server is running I think correctly as I'm running without issues an iotawatt posting correctly to them and also if I manual post inputs like:

xxxxxxxxx.xxxx.xx:8080/input/post.json?json={ct1: 0.15, ct2: 0.00, ct3: -0.00, ct4: -0.03, V: 1.01,psent:5,psuccess:0,freeram:25704}&node=esptest&apikey=xxxxxxxxxxxxxxxxxxxxxxxxxxxxx

(exactly the same as display on serial log from emonesp)
I see immediate input on my emoncms server.
I’ve no clue what can be wrong and already tested compile emonesp with esp8266 lib 2.4.2; 2.4.0; 2.3.0 always compile correctly upload and can see webpages, configure, etc… but never a connection to emoncms server

jeremypoulter, 2018-09-14:
If there anything set it the 'SSL SSH-1 Fingerprint:' field? If so remove that.

jeremypoulter, 2018-09-14:
No, ignore that, not the issue

E-NINA, 2018-09-15:
Hello, today I spent a litle bit more time on my issue and I'm preaty sure I've no problem with my emoncms server side. As i mentioned before manually typing posts data.
I end up with error "400"

/input/post.json?json={ct1: 615.29, ct2: 517.25, ct3: 430.61, ct4: 383.29, V: 310.32,psent:2,psuccess:0,freeram:25368}&node=emonesp&apikey=xxxxxxxxxxxxxxxxxxxxxx
Plain old HTTP
Emoncms error: server error: 400

Looking on "ESP8266HTTPClient.h" seems is a bad request

 HTTP_CODE_BAD_REQUEST = 400, 

On that file mentions different ways to make request and seems emonesp uses 1st (see below), 2nd should be the one where we can define port number. I'm not a programmer I'm not able to change code (I tried already without sucess ...) 

....
 // Plain HTTP connection, unencrypted
    bool begin(String url);
    bool begin(String host, uint16_t port, String uri = "/");
......

closed 2020-09-28 by glynhudson

dudman, 2021-02-12:
I have simular problem, to me it seems that the host can't have a port added to it because in http.cpp the port 80 is passed through to get_http_internal() and isn't parsed out and sent. Simplest I think would be to add into the interface an adition GUI field <PORT> which would default to 80 but can be changed.
I'll just try and hard code 81 instead of 80 and see if that works for me.
```
// -------------------------------------------------------------------
// HTTP GET Request
// url: N/A
// -------------------------------------------------------------------
String get_http(const char *host, String &path){
  WiFiClient client;
  return get_http_internal(client, host, path, 81, false);
} // end http_get
```

roscoegray, 2021-04-14:
Running into this issue also, as an add-on has been created for Home Assistant, having the ability to define the port would be great, and only if it omitted default to port 80. Originally raised an issue with [hassio-addon-emoncms](https://github.com/inverse/hassio-addon-emoncms/issues/13), but it would be better dealt here i would think.

qm3ster, 2022-01-18:
\> mfw
https://github.com/openenergymonitor/EmonESP/blob/0af0551c84d3e21b5d4161d634e49faba903c57d/src/http.cpp#L76
this should either be its own UI field or parse from `host:port`

qm3ster, 2022-01-18:
Fixed in https://github.com/openenergymonitor/EmonESP/pull/102