openenergymonitor/EmonESP #43: Factory reset of EEPROM data through GPIO0

danbates2, 2018-09-12:
If settings have been made on emonESP which make it difficult to use (wifi settings) or impossible to access (forgotten admin username and password) it would be very useful to enable a factory reset by holding down GPIO0 after boot.

I've seen this feature elsewhere on ESP devices. It's possible because the state of GPIO0 is only critical during boot, after which it can be used as a digital input, for example by initiating a factory reset after say, 5s of holding down.

jeremypoulter, 2018-09-12:
This is supported on the OpenEVSE WiFi firmware (which was based on EmonESP) and I am planning to merge all the changes from that code back to EmonESP. Will be resolved as part of that work. @glynhudson what do you think?

danbates2, 2018-09-12:
Looks like you've got it.
I cobbled this together if it helps.
https://github.com/danbates2/EmonESP/blob/master/src/src.ino

jeremypoulter, 2018-09-12:
Thanks that will do the job but stops the ESP doing anything while the Burton is pressed, also the LED is on GPIO0 so you have to be a bit clever if you want to use the LED for status.

FYI the OpenEVSE implementation is here: https://github.com/OpenEVSE/ESP8266_WiFi_v2.x/pull/180

danbates2, 2018-09-12:
A more elegant solution is better :)
It crossed my mind the LED could do some tripple blink or something indicating it's about to reset. What do you think?

danbates2, 2018-10-11:
I've made the quick fix, hold for 10s reset..
Unfortunately I've got to choose other priorities and don't have the time to implement your method Jeremy. Although it is better, I was having a headache trying to merge the relevant parts of wifi.cpp. I've made a [branch](https://github.com/danbates2/EmonESP/tree/openevse-implementation) on my fork to try and merge your method but have to leave it at that. I just reread your post about wanting to merge relevant changes back from OpenEVSE, so am trusting that 'as is' is good enough for now, to be revisited when time better suits.

closed 2018-10-11 by danbates2