openenergymonitor/EmonESP #50: Latest version (master) goes into bootloop on WemosD1 Mini

kerk1v, 2018-11-23:
Hello, 

The latest version built from dev goes into a boot loop on a Wemos D1 mini. 

Version 2.3.3  works fine

Both versions will report SSIDs found on the serial console, but none will be shown in the web gui to join, I had to enter mine manually into config.cpp, after which it worked fine. 

If you need help debugging on this hardware, please let me know,

sindudas, 2018-11-23:
May be you need to check that:

https://github.com/openenergymonitor/EmonESP#troubleshooting-upload

kerk1v, 2018-11-23:
> May be you need to check that:
> 
> https://github.com/openenergymonitor/EmonESP#troubleshooting-upload

First thing I did, and that´s not it. As I said, 2.3.3 doesn't bootloop (but doesn't show the Wlan SSIDs either)

sindudas, 2018-12-05:
Tried with a Wemos D1 mini the Master version, and it works. Shows all SSIDs found, select one and put the wifi password. Where did you get version 2.3.3? Latest release is 2.3.0.

closed 2020-01-29 by TrystanLea