openenergymonitor/EmonESP #58: Updater for Sonoff, flash size limitation.

danbates2, 2019-03-25:
The 1MB flash size on the sonoff makes it tricky to upload an update .bin file, as the updater I've found and integrated seems to require 50% free space.

Is there a way to update the flash with a .bin greater than 50% of the flash size? Off the top of my head, it would be risky because there's no way to guarantee the transfer of data over WiFi for the duration of the flash, so a sketchy wifi connection would soft-brick a sonoff.

@jeremypoulter does this sound right? It might be the case the updater is ditched for the sonoff.

danbates2, 2019-03-25:
@glynhudson @TrystanLea @jeremypoulter 

I put the src folder from emonesp/timer into the ESP8266_WiFi_v2.x/src folder and got the following:

DATA:    [=====     ]  53.7% (used 43956 bytes from 81920 bytes)
PROGRAM: [=         ]  10.5% (used 438603 bytes from 4194304 bytes)

The most interesting is the 81920 bytes. Where's the rest?

Getting rid of the spiffs code too would bring the size down a bit more. This seems like a good way forward to get below 0.5MB?

sindudas, 2019-03-25:
The Sonoff makes 2 firmware, sonoff-minimal and other sonoff-*
Sonoff-minimal is what is needed to use before update with other greater firmware.
And doesn’t use spiffs.

danbates2, 2019-03-25:
I can't find the minimal you refer to, is it a pio env?

sindudas, 2019-03-25:
On realeses, you can find the different .bin for the different purposes.

sonoff-minimal.bin = The Minimal version allows intermediate OTA uploads to support larger versions and does NOT change any persistent parameter. This version should NOT be used for initial installation.

When compiling from sources,there is a VAR for compiling that .bin .
That firmware handles the minimal functions to update it over web (at least)

danbates2, 2019-03-26:
I can find what look like 'minimal' memory addresses in firmware.map, but can't find the VAR you mention. Which file is it in?

sindudas, 2019-03-26:
[my_user_config.h](https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/my_user_config.h)
Line 435: //#define FIRMWARE_MINIMAL                         // Create sonoff-minimal as intermediate firmware for OTA-MAGIC

Uncommenting that line and compiling, generates the minimal firmware.

closed 2020-01-29 by TrystanLea