openenergymonitor/EmonESP #66: Compile error with Arduino IDE

wliva, 2019-10-09:
Hi,
I'm trying to upload EmonESP firmware with Arduino IDE to a NodeMCU ESP-12E V3 but I get some errors that abort the process.

The following are the installed software versions: 
-Arduino IDE 1.8.10
-esp8266 v. 2.4.2 (with 2.5.2 there are no upload errors but the board is not working)
-all libraries are loaded with the latest version

These are the errors:

libraries\ESPAsyncWebServer-master\WebServer.cpp.o: In function `std::__detail::_Nfa::_M_insert_subexpr_end(unsigned int, std::function<void (std::__detail::_PatternCursor const&, std::__detail::_Results&)> const&)':

C:\Users\Walter\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/StringArray.h:75: undefined reference to `std::__throw_regex_error(std::regex_constants::error_type)'

libraries\ESPAsyncWebServer-master\WebServer.cpp.o: In function `std::__detail::_Scanner<char const*>::_M_eat_escape()':

C:\Users\Walter\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/StringArray.h:75: undefined reference to `std::__throw_regex_error(std::regex_constants::error_type)'

C:\Users\Walter\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/StringArray.h:75: undefined reference to `std::__throw_regex_error(std::regex_constants::error_type)'

libraries\ESPAsyncWebServer-master\WebServer.cpp.o:(.text._ZNSt8__detail8_ScannerIPKcE10_M_advanceEv[_ZNSt8__detail8_ScannerIPKcE10_M_advanceEv]+0x6e): undefined reference to `std::__throw_regex_error(std::regex_constants::error_type)'

libraries\ESPAsyncWebServer-master\WebServer.cpp.o: In function `std::__detail::_Scanner<char const*>::_M_scan_in_bracket()':

C:\Users\Walter\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/StringArray.h:75: undefined reference to `std::__throw_regex_error(std::regex_constants::error_type)'

libraries\ESPAsyncWebServer-master\WebServer.cpp.o:C:\Users\Walter\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/StringArray.h:75: more undefined references to `std::__throw_regex_error(std::regex_constants::error_type)' follow

libraries\ESPAsyncWebServer-master\WebServer.cpp.o: In function `std::_Function_base::_Base_manager<std::__detail::_EndTagger<char const*, std::regex_traits<char> > >::_M_init_functor(std::_Any_data&, std::__detail::_EndTagger<char const*, std::regex_traits<char> >&&, std::integral_constant<bool, false>)':

C:\Users\Walter\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/StringArray.h:75: undefined reference to `std::__throw_bad_cast()'

libraries\ESPAsyncWebServer-master\WebServer.cpp.o: In function `std::__detail::_Compiler<char const*, std::regex_traits<char> >::_M_atom()':

C:\Users\Walter\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/StringArray.h:75: undefined reference to `std::__throw_bad_cast()'

C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lib\libstdc++.a(istream-inst.o): In function `__check_facet<std::ctype<char> >':

/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/streambuf_iterator.h:113: undefined reference to `std::__throw_bad_cast()'

C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lib\libstdc++.a(istream-inst.o): In function `std::istream::operator>>(int&)':

/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/streambuf_iterator.h:113: undefined reference to `std::__throw_bad_cast()'

C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lib\libstdc++.a(locale-inst.o): In function `std::ctype<char> const& std::use_facet<std::ctype<char> >(std::locale const&)':

/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/locale_classes.tcc:114: undefined reference to `std::__throw_bad_cast()'

C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lib\libstdc++.a(locale-inst.o):/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/locale_classes.tcc:114: more undefined references to `std::__throw_bad_cast()' follow

C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lib\libstdc++.a(ios-inst.o):(.literal._ZNSt9basic_iosIcSt11char_traitsIcEE5clearESt12_Ios_Iostate[_ZNSt9basic_iosIcSt11char_traitsIcEE5clearESt12_Ios_Iostate]+0x4): undefined reference to `std::__throw_ios_failure(char const*)'

C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lib\libstdc++.a(ios-inst.o): In function `std::basic_ios<char, std::char_traits<char> >::clear(std::_Ios_Iostate)':

/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_ios.tcc:122: undefined reference to `std::__throw_ios_failure(char const*)'

Multiple libraries were found for "Hash.h"
C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lib\libstdc++.a(c++locale.o):(.literal._ZNSt6locale5facet18_S_create_c_localeERPiPKcS1_+0x4): undefined reference to `std::__throw_runtime_error(char const*)'

 Used: C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\Hash
Multiple libraries were found for "ESP8266WiFi.h"
 Used: C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WiFi
Multiple libraries were found for "ESP8266HTTPClient.h"
 Used: C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266HTTPClient
Multiple libraries were found for "ArduinoOTA.h"
C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2/tools/sdk/lib\libstdc++.a(c++locale.o): In function `std::locale::facet::_S_create_c_locale(int*&, char const*, int*)':

 Used: C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ArduinoOTA
Multiple libraries were found for "ESPAsyncTCP.h"
 Used: C:\Users\Walter\Documents\Arduino\libraries\ESPAsyncTCP-master
Multiple libraries were found for "ESPAsyncWebServer.h"
 Used: C:\Users\Walter\Documents\Arduino\libraries\ESPAsyncWebServer-master
Multiple libraries were found for "ESP8266httpUpdate.h"
 Used: C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266httpUpdate
Multiple libraries were found for "EEPROM.h"
/Users/igrokhotkov/e/ESPTools/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/xtensa-lx106-elf/libstdc++-v3/src/c++98/c++locale.cc:207: undefined reference to `std::__throw_runtime_error(char const*)'

 Used: C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\EEPROM
collect2.exe: error: ld returned 1 exit status

Multiple libraries were found for "PubSubClient.h"
 Used: C:\Users\Walter\Documents\Arduino\libraries\PubSubClient
Multiple libraries were found for "ESP8266mDNS.h"
 Used: C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266mDNS
Multiple libraries were found for "DNSServer.h"
 Used: C:\Users\Walter\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\libraries\DNSServer
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

I think there is a problem with the async library.
How can I solve this problem?

Thanks in advance for your help.

danbates2, 2019-10-12:
In the board selection dropdown, can you try 'Generic ESP8266 Module' and report back?

fillycheezstake, 2019-11-08:
Just FYI, 

I see this user also had trouble when using ESP8266 for Arduino IDE version 2.5.2. 

None of my devices get an IP address from the EmonESP when I erase and flash the ESP fresh (so EmonESP goes into the AP mode). 

It looks like something is breaking DHCP address handouts in AP mode when using 2.5.2.  I'm able to roll back to 2.5.1 and everything works correctly.

glynhudson, 2020-09-28:
Recommend using PlatformIO to compile: https://guide.openenergymonitor.org/technical/compiling/

closed 2020-09-28 by glynhudson