openenergymonitor/EmonESP #69: SPIFFS log error en version 3.0.0

Stepanhuge, 2020-02-13:
When I download the firmware.bin and spiffs.bin from the last release (V3.0.0) and the esp creates a AP the html output that I can read from the device which is connected to espwifi_xxxx network the following error:

**/home.html not found, have you flashed the SPIFFS?**

Meanwhile with the previous version (V2.3.0 | Misc Bug Fixes) everything works fine, so I think that the spiffs.bin is the problem and I dont figure out how to compile by my selft to confirm the issue.

Best regards

NGrant92, 2020-03-18:
Any progress on this? I'm having the same issue. 

- I compiled from master with `pio run`.

- I then tried using the OTA command `pio run -t upload --upload-port <LOCAL-ESP-IP-ADDRESS>`

- When I go to the AP homepage I also got 
```
/home.html not found, have you flashed the SPIFFS?
```

- I tried to follow the [README.md command](https://github.com/openenergymonitor/EmonESP#c-ota-upload-over-local-network) to upload the spiffs file but I got the following error:

```
pio run -e emonesp_spiffs -t upload --upload-port <LOCAL-ESP-IP-ADDRESS>
Error: Unknown environment names 'emonesp_spiffs'. Valid names are 'hpmon, smartplug, esp12e, emonesp, wifirelay'
```

Stepanhuge, 2020-03-19:
1. Upload src via command “pio run -t upload”
2. Upload spiffs via command “pio run -t uploadfs”

if the issue persist try to upload the BLANK.bin to the memory directions before to try again

NGrant92, 2020-03-19:
> 2. Upload spiffs via command “pio run -t uploadfs”

I tried that and I hit another issue, the UI looked dated and it wasn't reading the values correctly.

I made a more detailed post about it in the [OEM forums](https://community.openenergymonitor.org/t/emonesp-firmware-upload-not-enough-space/13621)

NGrant92, 2020-03-19:
also how did you connect the emonESP device to your machine?

glynhudson, 2020-09-28:
Version V3.1.0 does not require SPIFFS upload. Only a single flash upload is required.

closed 2020-09-28 by glynhudson