openenergymonitor/EmonESP #70: [ERROR]: Bad Answer: ERR: ERROR[4]: Not Enough Space

NGrant92, 2020-03-19:
Hi,

I've tried to compile and upload from both the [master branch](https://github.com/openenergymonitor/EmonESP) and the [latest release](https://github.com/openenergymonitor/EmonESP/releases/tag/3.0.0). I followed the README.md the OTA instructions to compile and upload for both [PlatformIO](https://github.com/openenergymonitor/EmonESP#option-1-compile-using-platformio) and [ArduinoIDE](https://github.com/openenergymonitor/EmonESP#option-2-compile-using-arduino-ide). 

PlatformIO version: 4.2.1
Arduino ESP boards version: 2.5.0

Both method result in the same "Not Enough Space" error message  

```bash
> pio run -t upload --upload-port <emonesp-local-ip>

Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead
Processing emonesp (platform: espressif8266@1.8.0; framework: arduino; board: esp01_1m)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 1.8.0 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 2.20402.4 (2.4.2) 
 - tool-espotapy 1.1.3 
 - tool-esptool 1.413.0 (4.13) 
 - tool-mkspiffs 1.200.0 (2.0) 
 - toolchain-xtensa 1.40802.0 (4.8.2)
Converting src.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <PubSubClient> 2.6
|-- <ESP Async WebServer> 1.1.1
|   |-- <ESPAsyncTCP> 1.2.2
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <NTPClient> 3.1.0
|-- <ESP8266WiFi> 1.0
|-- <EEPROM> 1.0
|-- <ESP8266HTTPClient> 1.1
|   |-- <ESP8266WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266mDNS>
|   |   |-- <ESP8266WiFi> 1.0
|-- <ESP8266httpUpdate> 1.2
|   |-- <ESP8266HTTPClient> 1.1
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncTCP> 1.2.2
|   |-- <ESP8266WiFi> 1.0
|-- <Hash> 1.0
|-- <DNSServer> 1.1.0
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
Building in release mode
Compiling .pio/build/emonesp/src/src.ino.cpp.o
Retrieving maximum program size .pio/build/emonesp/firmware.elf
Checking size .pio/build/emonesp/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  45.8% (used 37488 bytes from 81920 bytes)
Flash: [========= ]  87.2% (used 435944 bytes from 499696 bytes)
Configuring upload protocol...
Looking for upload port...
Use manually specified:  <EmonESP-Local-IP>
Uploading .pio/build/emonesp/firmware.bin
16:59:50 [DEBUG]: Options: {'esp_ip': ' <EmonESP-Local-IP>', 'host_ip': '0.0.0.0', 'esp_port': 8266, 'host_port': 10531, 'auth': '', 'image': '.pio/build/emonesp/firmware.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
16:59:50 [INFO]: Starting on 0.0.0.0:10531
16:59:50 [INFO]: Upload size: 440096
Sending invitation to <EmonESP-Local-IP>
16:59:50 [ERROR]: Bad Answer: ERR: ERROR[4]: Not Enough Space
```

`Arduino Ctrl+U (port: <emonesp-local-ip>`
[image]

NGrant92, 2020-03-20:
It should also be noted that when I run pio run -t uploadfs --upload-port <EmonESP-Local-IP> it will upload spiffs.bin just fine.

HamzaHajeir, 2020-05-26:
I've the same issue, 

+ It's not understood, sometimes it just upload OTA fine in that big size, and sometimes it doesn't!

NGrant92, 2020-05-26:
> I've the same issue,

I forgot that I made this issue. I posted the same question on the [official forum](https://community.openenergymonitor.org/t/emonesp-firmware-upload-not-enough-space/13621/7) and had more help there.

Long story short, I had to upload through a wired connection instead of using OTA.

HamzaHajeir, 2020-05-26:
I see, 

What LD script you're using ? because for my case, your code size is OTAble.

here's my flags on platformio.ini : 
```
board_build.ldscript = eagle.flash.1m64.ld
build_flags = -D BEARSSL_SSL_BASIC
  -DNDEBUG
  -D PIO_FRAMEWORK_ARDUINO_LWIP2_HIGHER_BANDWIDTH_LOW_FLASH
```

1- Check whether it's OK to have just 64k SPIFFS.
2- Try this solution out.
3- Discuss whether `build_flags `are playing against your project or not. You can know by try-and-error. These flags reduces the code size anyway.

HamzaHajeir, 2020-05-26:
******** 

I'm not using EmonESP, I've just googled the error code and reached here.

glynhudson, 2020-09-28:
Please try the latest version V3.1.0, SPIFFS is not required

closed 2020-09-28 by glynhudson