openenergymonitor/EmonESP #77: Release 3.1.0 on SonOff Basic WIFI switch

stuartpittaway, 2020-09-29:
I'm trying to use the 1MB version of the emonesp on a sonoff basic wifi switch (no energy monitoring version),

The code flashes and runs, I can connect to the AP and get it to connect to my wifi, it has also been detected by DemandShaper in emoncms.

Sometimes the switch just reboots itself and flashes the green LED.  If I can get the device to be stable for a little while I can't save the settings from the control panel on the device.

Any ideas?

stuartpittaway, 2020-09-29:
Just to add, the device can be configured and appears to work correctly, it joins the home wifi network but after a few packets drops off and seems to reboot (flashing led).

```
Reply from 192.168.0.28: bytes=32 time=46ms TTL=255
Reply from 192.168.0.28: bytes=32 time=18ms TTL=255
Request timed out.
Request timed out.
Request timed out.
```

I'm thinking this is an EEPROM storage issue.

glynhudson, 2020-10-16:
What Sonoff are you using? The `smartplug` version works fine on the S20 plug, this doesn't have any energy monitoring. Try doing a full erase before upload to remove all parts of the old firmware. There is a precompiled .bin of the smartplug version  https://github.com/openenergymonitor/EmonESP/releases/download/3.1.1/smartplug.bin

stuartpittaway, 2020-10-16:
I've got the very basic version, this has hardwired connectors for the cables rather than a 3pin plug.

The code programs fine, just run into the problems above.

Onepamopa, 2022-08-27:
Have you been able to solve this? I've got a bunch of "Basic" switches that I need to operate via DemandShaper.
Still haven't tried flashing them.
Is there anything specific?