openenergymonitor/EmonESP #82: No ability to change the serial baudrate

Z0l, 2020-10-03:
The serial port in the normal use case with single phase monitoring only works on 115.200 and this value is hardcoded in the firmware; 3phase emonTX requires 9600 bps serial speed and that's the only reason 3phase owners need to compile custom firmware. It would be nice to have a checkbox to switch between the 2 serial speeds (or a dropdown list if there are other use cases which require a different value).

egoistdream, 2020-10-11:
> The serial port in the normal use case with single phase monitoring only works on 115.200 and this value is hardcoded in the firmware; 3phase emonTX requires 9600 bps serial speed and that's the [...]
Hi!

I have also compiled this firmware in order to change to the 9600bps serial spped but for me does not work.....
Have you managed to compile? I try using Ardruino and also platformio, the firmware was compiled but the serial do not work to send the data to the emon....

Regards

Z0l, 2020-10-11:
See the relevant forum topic here: https://community.openenergymonitor.org/t/unstable-emontx-emonesp-operation/14887/39
The baudrate is now in the debug.cpp not in src.ino, so you will have to change it there - EMONTX_PORT.begin(115200); to  EMONTX_PORT.begin(9600);
I compiled via platformio (also had to change the version of one dependency because it had trouble downloading it but I can no longer remember which one - you will see from the error message).