openenergymonitor/EmonESP #83: MQTT host IP address incorrect in debug

glynhudson, 2020-10-06:
The MQTT host IP address shown in the debug output is incorrect. See screenshot below, the correct IP address is 192.168.86.24, the correct address is being used by the ESP since the MQTT server is connected and working. However the IP address is slightly incorrect in the debug window: 

[image]

@jeremypoulter

jeremypoulter, 2020-10-06:
@glynhudson the right-hand is showing the wedserver requests so is the IP of the ESP not the MQTT server

glynhudson, 2020-10-06:
You're totally right 🤦

A user reported this, I should have looked at it properly. Sorry!

closed 2020-10-06 by glynhudson